psas/launch-tower #13: select LTC3 12-to-19 V DC-to-DC converter

nellump, 2015-02-03:
This converter will supply the rocket's shore power connection; 19 VDC @ ~ 4 A.

Start off by coming up with a list of promising candidates.  At this point, we will consider both board- and chassis-mount styles.

austinjoseph, 2015-02-13:
**This looked the most promising for a board mounted option, designed to run at constatn 3A out, not quite 4 but it looks like it might work.:
http://www.amazon.com/3-5-30V-4-0-30V-Booster-Converter-Regulator/dp/B008FLE7PA#productDetails

**Another like the above:
http://www.amazon.com/Converter-3-5-28V-1-25-26V-Adjustable-LM2596S/dp/B008ATU2X8/ref=pd_sim_e_1/178-7353896-8837034?ie=UTF8&refRID=0AZRG0X2AVXTCN79Z4CQ#productDetails

**Third option... still only 3A:
http://www.amazon.com/Retailstore-LM2596-Adjustable-Supply-Converter/dp/B009HPB1OI/ref=pd_sim_e_2?ie=UTF8&refRID=189ZVX2YQMF1KZNQXKFY

**This would require a bit of customizing but it has the right input -1 amp on the output but can put out 19V:
http://www.powerstream.com/Produz10.htm

**A comparison of similar converters:
http://www.powerstream.com/ADC-p006.htm

Finding no shortage of 10v to 20v but can't seem to find anything for 12v.
**Here is a 3 amp board mounted dip 10V to 12v DCDC converter:
http://www.globalspec.com/specsearch/partspecs?partId={3438B3BA-DC8B-4446-B89C-F950C086231E}&comp=3660&vid=125098&sqid=11131756

nellump, 2015-02-13:
China via eBay:
* "Waterproof 76W DC-DC Converter DC 12V to19V 4A Boost Step-up Car Power Module"
   * http://www.ebay.com/itm/Waterproof-76W-DC-DC-Converter-DC-12V-to19V-4A-Boost-Step-up-Car-Power-Module-/271772377430
   * 12 V in, 19 V @ 4 A out
   * Waterproof!!!!11
   * ~$13 shipped.

andrewgreenberg, 2015-03-30:
I *love* this last one. Sure, it's not PCB mounted, but you could mount it to the PCB if you wanted to, and then have some euroterminal blocks to screw it into nearby. Or you could mount it nearby on the case and do the same thing. I'm impressed with this one, especially if we buy 2-3 of them. Efficiency isn't outstanding, but it's rated to operate at industrial temperatures.

nellump, 2015-04-02:
I'll order a couple of those cheap Chinese switchers.  We can dissect them at a meeting to see how they look inside.

closed 2015-04-11 by nellump