psas/launch-tower #15: select LTC3 main battery cells

nellump, 2015-02-03:
Rechargeable LiPo cells.  14.4 VDC nominal; 4 AH.  See *Requirements* doc for details.

Same style as the existing ignition battery cells?

austinjoseph, 2015-02-04:
Finding tons of LiPo packs for 14.8 nom... hard to find datasheets.  Most of the 4-4.5 mAh run about $50-60.

-Multistar 4mAh 14.8v 4cell (at $24.58, it's the cheapest but still not in the 14.4 range)
 http://www.hobbyking.com/hobbyking/store/__56847__Multistar_High_Capacity_4S_4000mAh_Multi_Rotor_Lipo_Pack.html

-List of Lipo's
http://www.radicalrc.com/category/Lipos-3701--Up-539

-Informative website about LiPo
http://www.aircraft-japan.com/en/datasheet/polyquestxp.htm

nellump, 2015-04-11:
We really ought to perform a power survey of the LTC3 design in  order to determine a suitable battery capacity.  We need to consider the current coming in from the solar cells and/or lab power; the current consumption of all the major LTC3 components (BeagleBone, DC-DC converters, etc.); and the current consumed by any external devices.  Then we need to decide how long we want the battery to last given average and worst case load scenarios.

austinjoseph, 2015-04-18:
We researched the total theoretical worst case scenario power consumption based on the following:

Research:
	Maximum amount of time for battery draw on LTC lipos.
        Minumum amount of current provided.

To calculate the power requirements on the batteries
we specifically needed to calculate

power draw of shore power, BBB, sensors triggers on board.

Shore is 76 Watts
BeagleBone 5 Watts
Converters Worst case efficiency 70%  @  ? Watts
    *   12-5DC      7.14 Watts
    *   12-19DC	    27.14

115.28 Watts @ 14.4 V = ~ 8 amps

Methodology:

input power * efficiency = output power

All sensors are in the range of 400uA which we think the BBB should be able to handle.
Entire switching negligible as well (in the order of mWatts).

Continues Discharge Current on our Lipo pack will need to be ~8A.
Lipo Packs that can provide that current will be able to last an hour to an hour and a half and will cost roughly $120-150.

As the power consumption we found is twice as much as was expected in the data sheets we would appreciate some feedback on the methodology.

BartMassey, 2015-04-19:
Why is shore power 75W, or 30W for that matter? 10W continuous seems like pretty much the maximum you should ever need here unless I'm missing something major. Pointer to the LTC III document that describes power?

15W * 1.33 / 15V = 1.33A

I think this means that you can use a standard laptop battery ($20) and get several hours.

OutboundFlight, 2015-04-19:
We were just going based on the specs that Andrew gave us, which was 4A max at 19V, which we backed out to 80W total.

nellump, 2015-04-19:
On Sat, Apr 18, 2015 at 09:55:05PM -0700, Bart Massey wrote:
> Why is shore power 75W, or 30W for that matter? 10W continuous seems
> like pretty much the maximum you should ever need here unless I'm
> missing something major. Pointer to the LTC III document that
> describes power?

The LTC3 requirements call for a shore power supply of 19V at up to
4A.  The original shore power supply in LTC2 was only good for 2A, and
apparently that did not suffice.

-- 
Paul

aSmig, 2015-04-26:
Regarding shore power pulling more than 10 watts;  The FC alone can pull up to 14 watts.  Add two raspberry Pis, a couple STM32 Olimex boards, WiFi amp & transceiver, GPS board, IMU, RNH, servo... and a beefy battery charger.  It might never add up to 76 watts in nominal usage, but we do brown out on power up with less than 40 watts.  AV4 is embarrassingly not power optimized.  But I don't think low power was one of the design goals for AV4.  We did select an Atom, after all.

~8A @ 14V looks right to me.  14.8V is well within our 16.8V max and can probably be considered as good as 14.4V.

Instead of looking for larger capacity 14V packs, I think we should aim for a common size with a good price point and just stack them.  Also, we might get better DC-DC efficiency if we are switching down to 19V from 29V instead of buck-boosting up from 14V.  This would require some reworking of our DC-DC and PV charging systems.

OutboundFlight, 2015-04-26:
So, I was under the impression that the power requirement for the FC was already at 19V. That's what we heard from Andrew, and it's what's in our design spec.

OutboundFlight, 2015-05-01:
Based on discussion tonight we have a new proposal: use two 4-cell custom built LiPos in series, giving us a main battery voltage 28.8V. The main win from this is that we can use way cheaper DC-DC converters (~$30 instead of $100+) and use the same architecture as the Flight Computer. This allows us to copy the charge distributor board from the FC, and use the same battery architecture, so we can have extra spares lying around.

The part number for the LiPo battery cell we currently use is PL5567100, rated at 3.7V, 4250mAh.

Downconverters:
http://www.mouser.com/ds/2/405/slts228c-128240.pdf
We could use this TI converter for both the 5V and 19V rails.