psas/launch-tower #16: select LTC3 ignition battery charger

nellump, 2015-02-03:
12 VDC input, direct from PV panels.  Not integral to the LTC main board; chassis-mount only.  See *Requirements* doc for details.

roman-ku, 2015-02-13:
The ignition battery charger would probably have to be turned on and off manually. I don't the exact details but probably in the field and only when needed.

LiPo battery chargers aren't cheap but it would be still be wise to spend a little more. At a minimum the LiPo charger should have balancing capabilities. A bonus would be temperature sensing, so LiPo charger can auto turn off when it hits a certain temperature threshold.

One option we could use: [Hobbyking® DC-4S Balance Charger & Cell Checker 30w 2s~4s ](http://www.hobbyking.com/hobbyking/store/uh_viewItem.asp?idProduct=36906)

nellump, 2016-09-14:
Ancient issue; no longer relevant.

closed 2016-09-14 by nellump