psas/launch-tower #17: Investigate BBB display options

nellump, 2015-02-03:
The LTC3 *Requirements* doc calls for a front panel display to convey various vital signs to the user.  Research what the BeagleBone Black supports, and shop for promising candidates.

austinjoseph, 2015-02-03:
-BeagleBoneBlack Cape ("4.3) LCD, only compatible with BBB... It's a touch screen and pricey at ~$65
     -https://www.sparkfun.com/products/12085
     -http://www.mouser.com/ProductDetail/4D-Systems/4DCAPE-70T/? qs=AAveGqk956GzboWsp1MGLg%3D%3D&kpid=1024863130&gclid=CJy2lre8xsMCFdKEfgodRBMAgw   (This is a 7 inch similar version for $89)

- "4.3 40-pin TFT Display ~$26... also an LCD
     http://www.ebay.com/itm/4-3-034-40-Pin-TFT-Display-480x272-With-Touchscree-n-/201111080157?_trksid=p2054897.l4275

austinjoseph, 2015-04-17:
Denistron has a line of outdoor lcd's [here][1] is a link to a typical data sheet.
   *   8.4" diagonal display size
   *   Anti-glare
   *   4 wire resistive touch
   *   "Vandal" resistant film to glass design.

[1]http://www.densitron.com/GetPdf.aspx?nDisplayID=489

nellump, 2016-09-14:
Ancient issue; no longer relevant.

closed 2016-09-14 by nellump