psas/launch-tower #18: Select Relay Driver Technology for LTC3 External Triggers

nellump, 2015-03-28:
The BeagleBone GPIO have a very limited current sourcing capability, not nearly enough to drive a typical 12V relay.  We'll either need to use discrete driver transistors, one each connected to its own GPIO, or use a relay driver IC (connected to the BBB over I2C or SPI).  Investigate amongst yourselves.  Refer to the Requirements Spec for details, as necessary.

nellump, 2015-03-28:
Linear Technology offers an I2C-based relay driver chip, [LTC4255][1].  Four independent open-collector drivers; short-circuit and over-current (foldback) protection; etc.

[1]: http://cds.linear.com/docs/en/datasheet/4255f.pdf

nellump, 2015-03-28:
Relay-driving capes are also available, including a [four-relay 3 A cape][1] from CircuitCo.

[1]: http://elinux.org/CircuitCo:Relay_Cape

nellump, 2015-03-29:
Possibilities:
* MC33879: Configurable Octal Serial Switch, Open Load Detect
  * http://www.freescale.com/webapp/sps/site/prod_summary.jsp?code=MC33879

BartMassey, 2015-03-30:
What are the relay coil current requirements? Why are we using relays instead of solid-state switches in the first place? Is there really any reason not to just use a MOSFET per GPIO?

nellump, 2015-04-01:
Haven't settled on a specific relay, yet.  IIRC, 30-50 mA is a typical 12 V relay coil current.

Relays seem simpler and more robust to me, but that could be due to lack of experience with MOSFETs.

OutboundFlight, 2015-04-01:
Isn't there also a safety/reliability issue? Because the relays are driving the igniter and RocketReady for the launch system. Are relays considered more reliable than transistors?

BartMassey, 2015-04-01:
Transistors are way more reliable than relays if they are protected properly. Mechanical failure modes are unpredictable, whereas transistor failure modes are highly predictable/repeatable.

Just use a MOSFET directly. If there are unusual voltage or current requirements, use a "solid-state relay" (http://www.digikey.com/product-search/en/relays/solid-state-relays/1048664). If you must use a mechanical relay use a MOSFET for the coil.

nellump, 2015-04-02:
The current implementation is based on your second suggestion (MOSFET driving a mechanical relay's coil).  (This is also Andrew's suggested circuit.)  The MOSFETs are good for several amps, though.  Would we need to worry about heat dissipation if we drove the external triggers directly with MOSFETs?

BartMassey, 2015-04-02:
I doubt it. The effective resistance of the MOSFET is really tiny, so it doesn't really dissipate noticeable power except when switching. Like I say, check out the "solid state relay" chips I referenced earlier: they are basically a FET with some protection circuitry. I think you'll find reasonably-priced things rated for our application.

andrewgreenberg, 2015-04-06:
You have to worry a little bit about thermal dissipation, but only if you choose the wrong one. Here's a [better list](http://www.digikey.com/product-search/en?FV=fff40010%2Cfff80058%2Ca40154%2Ca40155%2C1140003%2C1140050%2C1fc003e%2C1fc01ca%2C1fc01cb%2C1fc01cc%2C1fc01cd%2C1fc021c%2C1fc021d%2C1fc021e%2C1fc028d%2C1fc028e%2C1fc050e%2C1fc0515%2C1fc18f3%2C1fc1af6%2C1fc1b1e%2C20800ed%2C6c4001e%2C6c4001f%2C6c40023%2C6c40024%2C6c40047%2C6c40056%2C6c4006c%2C6c4006e%2C6c4007d%2C6c40086%2C6c40092%2C6c400d6%2C6c400de%2C6c80040%2C6c80041%2C6c80042%2C6c80043%2C6c80044%2C6c80045%2C6c80046%2C6c80047%2C6c80048%2C6c80049%2C6c8004a%2C6c8004b%2C6c8004c%2C6c8004d%2C6c8004e%2C6c8004f%2C6c80050%2C6c80053%2C6c80074%2C6c80075%2C6c80076%2C6c80077%2C6c80078%2C6c80079%2C6c8007a%2C6c8007b%2C6c8007c%2C6c80087%2C6c80098%2C6c80099&mnonly=0&newproducts=0&ColumnSort=1000011&page=1&stock=1&pbfree=0&rohs=0&quantity=&ptm=0&fid=0&pageSize=25) for you from Digi-Key. Some of them are absolutely huge, either in area or height, but many of them are simple 8 pin SMT parts.

Here's my favorite huge [ginourmous one](http://www.digikey.com/product-detail/en/CPC1907B/CLA394-ND/3908816).

And my favorite [small one]( http://www.digikey.com/product-detail/en/TLP3542%28F%29/TLP3542F-ND/871299).

I'd go for the small one. You'll still need a small N channel MOSFET to run on the LED of the "solid state relay", but that's OK. And now no bouncing of the contacts :)
 but no clicking noise either :(

We can chat Tuesday, but let's wrap it up then so you can move on.

nellump, 2015-04-09:
Andrew and the LTC3 team conferred at Tuesday's general meeting, and have settled on using MOSFET-driven photorelays for control of external devices' power and triggering.  I will update the design spec accordingly.

closed 2015-04-09 by nellump