psas/launch-tower #19: Select enclosure for LTC3

nellump, 2015-04-11:
Shop for a new enclosure for LTC3.  The redesign should allow us to use a significantly smaller enclosure than the one used by LTC2.

The requirements doc does not include many specific details about the enclosure.
   * Either heavy duty plastic (like LTC2) or metal would be acceptable.
   * Dust and dirt protection would be a very good idea.
   * Moisture-proof would be nice, but unlikely to ever be a problem in our launch environment.
   * The enclosure-to-tower mounting scheme needs to be sturdy and straightforward (tool-free, preferably).

We'll need to have a good idea of the size of the LTC3 main board prior to making a final selection.

austinjoseph, 2015-04-11:
NEMA Boxes has this enclosure.
   *   Gasketed
   *   Has two doors, the first is clear plexiglass the second is aluminum.
   *   Easy clip for access for possible front LED.
   *   They make an outdoor version that is UV stabilized.
   *   Dim offered are: 
     *   11.3X7.5X5.5 (H/W/D rough)
     *   15.7X11.7X6.3 (H/W/D rough)
     *   19.2X14.6X7.9 (H/W/D rough)
   *    http://www.budind.com/images/product/large/nema-boxes-uni-nbe.jpg

austinjoseph, 2015-04-15:
Hammond offers [this enclosure][1] in their "NEMA Boxes" line:
   *   Gasketed
   *   Mild steel has UV protected cutout for LCD
   *   Flanged mounts available.
   *   Dim offered are: 
     *   Highly variable.
     *   16X12X6 (H/W/D rough)
        *   Viewing Window:  11.33X6.71(H/W)

[1]: https://www.hammfg.com/electrical/products/industrial/en4sdw

austinjoseph, 2015-04-15:
Hammond also offers [this enclosure][1]:
   *   Gasketed
   *   Mild steel.
   *   Flanged mounts.
   *   Tooled lock
   *   Galvanized removable panel
   *   No front panel for LCD
   *   Dim offered are: 
     *   Highly variable.
     *   16X12X6 (H/W/D rough)
        *   Viewing Window:  11.33X6.71(H/W)

[1]: https://www.hammfg.com/electrical/products/industrial/ej

nellump, 2015-04-15:
We'll need to consider how the LTC enclosure will mount to the tower.  Many (most?) of the industrial enclosures we've seen support DIN mounting rails, which are pretty slick.  [Read][1] and [watch][2] all about it.

[1]:http://www.designnews.com/author.asp?section_id=1386&doc_id=224061&print=yes
[2]: https://www.youtube.com/watch?v=MuF49GPcNG4

natronics, 2015-04-15:
How much do these boxes weigh?

austinjoseph, 2015-04-15:
3-16lbs

dsigma, 2015-04-16:
The current plan is to mount a piece of 0.25" aluminum flat bar across the back of the tower, and you can bolt/drill/screw anything you want to that flat bar. However, we want to avoid things protruding too far from the flat bar so that the sections can lay nicely on the ground and avoid snagging people while moving them.

austinjoseph, 2015-04-16:
They all have or have the option to come from the factory with a flat
angled bar that comes off the top side of the enclosure.