psas/launch-tower #22: Create schematic sheet for LTC3 power in

nellump, 2015-04-18:
A placeholder sheet ("Power In") already exists. This schematic should contain circuitry related to the solar cell and lab power inputs (connectors, conditioning, etc.).  Refer to the design spec and LTC2 schematics for guidance.

Please do not commit your work to the repo's "master" branch. Create a new branch. We can merge to "master" when your're finished.

austinjoseph, 2015-04-20:
Taking a shot at this.  Are we just going to use pins for the power in?  I'm looking at other options (XLR) because it seems like it would be nice if things were uniform and easily plugged in unplugged.  Thoughts?

nellump, 2015-04-20:
On Mon, Apr 20, 2015 at 07:25:26AM -0700, austinjoseph wrote:
> Taking a shot at this.  Are we just going to use pins for the power
> in?  I'm looking at other options (XLR) because it seems like it
> would be nice if things were uniform and easily plugged in
> unplugged.  Thoughts?

Unless we want to mount the connectors directly to the board, and then
deal with mounting the board in the enclosure so that it's flush with
the sides (all of which sounds like extra work), we should probably go
with something generic like Eurostyle connectors on the board, which
can then be wired up to the external connectors mounted on the
enclosure.

nellump, 2016-09-14:
Ancient issue; no longer relevant.

closed 2016-09-14 by nellump