psas/launch-tower #23: LTC Architecture overview - Review switching from 4s to 8s batteries, plus part choices

OutboundFlight, 2015-05-09:
Cross-posted to Avionics list.

I wanted to send out an update about some design decisions the LTC made recently and solicit feedback.

Based on internal discussions and conversations with Andrew and Kenny, we've decided to swtich from a 14.4V nominal, 4s lipo battery to a 28.8V nominal, 8s lipo configuration.

The primary advantage of this is that it lets us use buck converters down to our power rails at 19V and 5V; configurable buck converters are cheap and easy to find, and have higher efficiencies at a low price point than the boost converters.

The downside of this is that we need to do a custom charging circuit because we're moving away from standard 12V nominal MPPT and 4s battery setups.

But, upon further investigation, we found the LT8490, which is a combination charger/voltage regulator/solar MPPT, and is discussed below. It will be difficult to implement but is a big win as far as cost and system flexibility.

PRELIMINARY COMPONENT CHOICES FOR LTC POWER SYSTEMS:
Voltage regulators:
This is a 7-pin drop-in converter. It's about $30 per, but it will just work, and give us the current, power, and temperature specs we need.
TI PTN78020H (19V rail)
TI PTN78020W (5V rail)

Battery front-end:
LT8490
This is a switching buck/boost regulator with a wide input/output range + charging (LiPO compatible) + solar MPPT. It's very robust and costs about $15. This will be the most complicated circuit and largest time investment of our design, but the work will largely transfer over to the CubeSat power systems.

Battery protection:
TI BQ77PL900
This is a cell protector/balancer that normally runs on I2C with an analog front-end, but can be configured for stand-alone operation. It is cheap, relatively simple to use, and will do cell balancing for us in the field.

Battery:
Because we need at least a 2V overhead to provide 19V for shore power in order to use a down-regulator, we will likely use an 8s configuration. 6s is too close (21.6V) to the output, and 7s is available but is a very non-standard battery configuration. 
Based on assumptions: 90% efficiency, calc. 96W of power (a conservative estimate), 4 hours pure battery, we need a capacity of:

13.2Ah at 28.8V nominal (8s)
15.2Ah at 25.2V nominal (7s)
17.8Ah at 21.6V nominal (6s)

We can get two 8s packs at 5.8Ah or four 4s packs at 5.8Ah for around $160.
Here's a good option:
http://www.hobbyking.com/hobbyking/store/__21385__ZIPPY_Compact_5800mAh_8S_25C_Lipo_Pack.html

Here's a higher energy-density, lower-discharge (but more than high enough for our purposes) option for the same price:
It's a 4p2s pack with 16Ah of storage.
http://www.hobbyking.com/hobbycity/store/uh_viewItem.asp?idProduct=66309&aff=3019
However, it seems to be perpetually out of stock. We'll keep an eye on it; this would be bulkier but a ton of storage.

OutboundFlight, 2015-06-25:
Given that we seem to be going ahead with this design, I'll close this issue.

closed 2015-06-25 by OutboundFlight

AichaPacem, 2016-05-16:
Hi,
Is there any update concerning LT8490 schematics? :)