psas/launch-tower #29: Find/Create footprint for LTC2991

austinjoseph, 2015-09-30:


nellump, 2015-09-30:
I created both a schematic symbol and PCB footprint for the LTC2991 when I designed a break-out board for it.  They're available in my kicad-libs repo:

Symbols:
   * https://github.com/nellump/kicad-libs/tree/master/symbols
   * Get both LTC299x.dcm and LTC299x.lib

Footprint:
   * https://github.com/nellump/kicad-libs/blob/master/footprints.pretty/MSOP-16.kicad_mod

nellump, 2015-10-01:
The LTC3 schematics already include a pair of LTC2991 ICs.  I've added a copy of the LTC2991.dcm and .lib files I created for the breakout board, so we have our own copy here (in addition to whats in Launch_Tower_Computer_III-cache.lib).

I've also added a copy of the MSOP-16 footprint, so it's there when we're ready for it.

closed 2015-10-01 by nellump

austinjoseph, 2015-10-02:
Ah cool, thanks Paul.  Was just going through stuff and seeing what I could
get done.  The footprint looks great.  We are going to do some
administrative housekeeping today and create/close git issues.  Are you
back in town yet?